Entware-for-kernel-3x/Entware-ng-3x #1: install_std.sh script, and libc corruption

amurabito, 2016-08-24:
after sourcing the install.sh from 
http://entware-3x.zyxmon.org/binaries/armv7/installer/install_std.sh

there appeared to be an issue with attempting to get a handle: 

wget: can't open '/opt/lib/ld-2.23.so': Text file busy

this ended up nuking libc, which as I'm sure you can imagine, has caused me some trouble.

/opt/lib/libc.so.6: file too short

any tips on attempting to replace libc? I still have ssh access, however most all shell commands (dependent on libc) cease to work.

amurabito, 2016-08-24:
for some more details: ARMv7, on a Netgear R7000. Existing entware-ng in place, on top of TomatoUSB

amurabito, 2016-08-24:
I may have a fix, through simply unmounting the USB drive, and manually attempting to fix the files with another system. I will try this today. 

If I have used your installer script incorrectly, feel free to let me know, and I'll update the documentation.

amurabito, 2016-08-24:
i have successfully resolved the corrupted libc with the use of another system. I suppose this issue should track that the shell script has the possibility to fail, and that failure may affect use of the shell.

zyxmon, 2016-09-27:
Not enough information to track installation issue, no installation log. I'll reopen the issue if there will be more similar cases.

closed 2016-09-27 by zyxmon