Entware-for-kernel-3x/Entware-ng-3x #7: Alt install stalls at 45% with QTS 4.3.3

stecca21, 2017-02-16:
I'm not sure if this is due to this version of QTS being released today or if something else is wrong, but when I try to manually install the alt package the install stalls at 45%. This does not happen when the standard install, but the standard install wipes /opt on reboot and I don't want that.

zyxmon, 2017-02-17:
Please remove the standard package. Download alt package to your NAS and install it manually by running
`sh <path_to_Entware-3x_0.98alt.qpkg>`. Please post the output here. I have installed QPKG many times on different NASes with no problems.

zyxmon, 2017-02-17:
Entware uses /opt folder by design. If you want to install other packages that use opt - do not install Entware. Entware is used on many different embeded devices not only on Qnap.

stecca21, 2017-02-17:
I understand this. I think you misunderstood me.

I lost all my packages on reboot with the standard installation. This is
why i need the alt install to work.

Allison Stec

zyxmon, 2017-02-17:
You are to find the reason why your /opt folder is lost. My guess is that some qpkg is doing it. As somebody stated on qnap forum - antimalware is blocking entware.
I do not see the installation log from you.

stecca21, 2017-02-18:
The alt package installs using sh but fails if i use the package manager to install.

stecca21, 2017-02-18:
Also, if you could include a link to the post you've seen on the QNAP fourm that says antimalware is causing the wiping of /opt on reboot I'd appreciate it. I can't seem to find that.

zyxmon, 2017-02-18:
I cannot confirm that alt package fails when App Center is used to install Entware downloaded as a file. But I do confirm that Entware does not run after reboot. HD_Station interferes and also uses /opt.
I'll try to find a way for Entware to coexist with HD_Station.
Just now you can start Entware with `..../.qpkg/Entware-3x/Entware-3x.sh start` command. The location of   `qpkg` folder depends on your setup!

zyxmon, 2017-02-19:
@stecca21 - the solution is here 
https://forum.qnap.com/viewtopic.php?f=351&t=130320

Please ask QNAP specific question on qnap forum.

closed 2017-02-19 by zyxmon