Entware-for-kernel-3x/Entware-ng-3x #8: nginx doesn't work anymore.

strichking, 2017-02-26:
nginx stop working since last release. I was in 1.10.1-2(with dd-wrt r25697) and then moved to 1.10.2-1( dd-wrt r31277). Everything was working OK with the old release. My router model is tl-wr1043nd.
I made a rollback for the last stable version of dd-wrt but it doesn't work neither.
This is printed when I run ./S80nginx start
nginx: [emerg] getpwnam("nobody") failed (2: No such file or directory) in /opt/etc/nginx/nginx.conf:1
Tried changing from nobody to everyone/root and nothing works.

zyxmon, 2017-02-27:
I have just tested nginx with 2 users that exist on my system. It works.
What error nginx displays when starting manually (not from the script)? Probably one of the libs needs to be reinstalled. opkg system is not clever enough do do it.

strichking, 2017-02-28:
Yes, you are right. I reinstalled the last dd wrt beta version and started opkg script. It installed flawlessly, nginx too, but there is something wrong with stability since transmission crashes randomly and it even doesn't start. I noted something with the old version of dd-wrt, it crashes when opkg is running the creation of locale files. The old version is the last stable version. r25697.

zyxmon, 2017-02-28:
I am closing the ticket - nginx works.

As for transmission - you can try to use an old version from archive folder or anothe torrent client like rtorrent or deluge.

closed 2017-02-28 by zyxmon