Entware-for-kernel-3x/Entware-ng-3x #12: libz error after restart

tautomer, 2017-03-21:
I am trying to install transmission on my ea8500 running on dd-wrt r31700.

Everything was successful, but neither opkg nor transmission could run properly after reboot. I added rc.unslung

Delete everything under /jffs/opt/ and reinstall can solve the problem.

This time I got libz.so.1 error. 
[image]

I also received some other errors, but all related to library files or segmentation fault. 
[image]

It looks like mostly environment problem. How can I fix this one? What else can I provide for diagnostics?

zyxmon, 2017-03-21:
Entware developers do not have routers running DD-WRT. We suggest to solve your problem on DD-WRT forum. Transmission is running perfectly on ARMv7 hardware:

```
# opkg install transmission-daemon-openssl
Installing transmission-daemon-openssl (2.92+git-4) to root...
Downloading http://entware-3x.zyxmon.org/binaries/armv7/transmission-daemon-openssl_2.92+git-4_armv7-3x.ipk
Installing libevent2 (2.0.22-1) to root...
Downloading http://entware-3x.zyxmon.org/binaries/armv7/libevent2_2.0.22-1_armv7-3x.ipk
Configuring libevent2.
Configuring transmission-daemon-openssl.
# /opt/etc/init.d/S88transmission start
 Starting transmission-daemon...              done.
# ps | grep transm
11443 root     32264 S    transmission-daemon -g /opt/etc/transmission
11484 root      3052 S    grep transm
```

closed 2017-03-21 by zyxmon

tautomer, 2017-03-22:
I tried some different version of dd-wrt. r31221 works well, so I am just running on it.

Not sure what happened to the new versions.

zyxmon, 2017-03-22:
@tautomer 
In new DD-WRT versions wget downloads corrupted files. Once Entware-3x is installed and /opt/bin is before /bin in the PATH variable wget is used from Entware-3x and works. Read more here https://github.com/Entware-ng/Entware-ng/issues/595