Entware-for-kernel-3x/Entware-ng-3x #14: opkg update - segmentation fault

emitar, 2017-04-01:
Hi,
I have DD-WRT v3.0-r31722 std on Archer C5

```
uname -a
Linux DD-WRT 3.18.48 #2573 Tue Mar 21 05:53:25 CET 2017 mips DD-WRT
```
```
 df -h
Filesystem                Size      Used Available Use% Mounted on
rootfs                   12.0M     12.0M         0 100% /
/dev/root                12.0M     12.0M         0 100% /
/dev/sda1               476.9M    382.0K    447.0M   0% /opt
/dev/sda3                 2.9G     16.0E      2.8G 100% /tmp/mnt/sda3
```

And have problem with instalation using 'install_std.sh':
```
sh ./install_std.sh
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Info: Opkg package manager deployment...
Connecting to entware-3x.zyxmon.org (81.4.123.217:80)

Connecting to entware-3x.zyxmon.org (81.4.123.217:80)
opkg.conf            100% |*******************************|   147   0:00:00 ETA
Connecting to entware-3x.zyxmon.org (81.4.123.217:80)

Connecting to entware-3x.zyxmon.org (81.4.123.217:80)

Connecting to entware-3x.zyxmon.org (81.4.123.217:80)
Connecting to entware-3x.zyxmon.org (81.4.123.217:80)
Info: Basic packages installation...
**Segmentation fault
Segmentation fault
Segmentation fault**
Info: Congratulations!
Info: If there are no errors above then Entware-3x was successfully initialized.
Info: Add /opt/bin & /opt/sbin to your PATH variable
Info: Add '/opt/etc/init.d/rc.unslung start' to startup script for Entware-3x services to start
Info: Found a Bug? Please report at https://github.com/Entware-for-kernel-3x/Entware-ng-3x/issues
```

I try to find solution but no result. Any idea?

zyxmon, 2017-04-01:
It is DD-WRT problem described here - https://github.com/Entware-ng/Entware-ng/issues/595

closed 2017-04-01 by zyxmon

treii28, 2017-09-26:
ok, and....? what do you do to fix it?

zyxmon, 2017-09-26:
The latest DD-WRT firmwares do not have this bug.