Entware-for-kernel-3x/Entware-ng-3x #15: iptables-save on dd-wrt

Xentrk, 2017-04-06:
DD-WRT has an older version of iptables. There are also issues with iptables-save and iptables-restore commands not working on DD-WRT builds and there are forum postings on dd-wrt forums about this.  I was hoping to resolve it by installing entware package iptables from this repository.  I ran the updated malware-filter script (links below).  iptables-save commands results in

Can't find library for target `TRIGGER'

In /opt/sbin, the ls -l command shows this:

ip6tables-save -> xtables-multi

My path is as follows:
PATH=/opt/sbin:/opt/bin:/bin:/bin:/usr/bin:/sbin:/usr/sbin:/jffs/usr/sbin:/jffs/usr/mmc/sbin:/mmc/bin:/mmc/usr/sbin:/mmc/usr/bin

Thanks you in advance for the help.

Malware-Filter
https://github.com/RMerl/asuswrt-merlin/wiki/Ipset-script-installation-instructions#malware-filter
code: https://gitlab.com/swe_toast/malware-filter/raw/master/malware-filter

zyxmon, 2017-04-06:
iptables package in entware was built for systems that do not have iptables. It is known to work it such cases. Some iptables commands need corresponding kernel modules or `opkg install xtables-addons_legacy` with corresponding kernel modules.
Please use iptables from your firmware.

closed 2017-04-06 by zyxmon

Xentrk, 2017-04-06:
Thanks zyxmon. Unfortunately, iptables-save and iptables-restore command is not part of DD-WRT builds.  The builders decided to not include those commands in order to save space is the reason listed in the forums.  I was able to get the malware-filter script to work by replacing iptables-save with iptables -L.  I don't see the package you mentioned on http://entware-3x.zyxmon.org/binaries/armv7/Packages.html