Entware-for-kernel-3x/Entware-ng-3x #17: about libtiff

wdj315, 2017-04-26:
### Package: libtiff 4.0.6

For existing package:

- expected behavior: ...command like tiffcp  can run
- actual behavior: ...  all command can not run ...
my device is QNAP NAS 231+ 

For new package to be added:

- make sure there's no similar package already added,
- describe killer features of new package,
- make sure it's open source and not abandoned (at least for last 5 years or so),
- please, use native compilation for additional PHP, Perl, Ruby, Python modules for non-MIPS platforms.

### Platform:  armv7

- please, tell us what feed and\or devices are you using.
- be ready to check issue-related  changes or it will be closed after 30 days with "no response" status.

zyxmon, 2017-04-26:
tiffcp and other binaries are in tiff-utils package, just run `opkg install tiff-utils`.

closed 2017-04-26 by zyxmon

wdj315, 2017-04-26:
OMG!  so simple = =
thank you very much, I should ask you earlier .....
it bother me 3 days ><
You are the man!