Entware-for-kernel-3x/Entware-ng-3x #18: dropbear pid file

baragoon, 2017-05-12:
### Package: dropbear - 2016.74-2

For existing package:

- expected behavior: file /opt/var/run/dropbear.pid containing running process PID
- actual behavior: can't open '/opt/var/run/dropbear.pid': No such file or directory

### Platform:

Android TV box, aarch64, Amlogic x96 and Amlogic em95x. Both boxes having the same behaviour.

zyxmon, 2017-05-12:
Entware is experimental on android. Everything is OK with dropbear and dropbear pid on other devices.

If your android box uses 3.x. or higher kernel - Entware-3x (use alternative install) is recommended.

closed 2017-05-12 by zyxmon

baragoon, 2017-05-12:
This is strange but if i manually killed dropbear processes and start it again with /opt/etc/init.d/S51dropbear start pid file is created successfully.