Entware-for-kernel-3x/Entware-ng-3x #20: libstdbuf.so not found on DD-WRT/Arm7

nfitzkee, 2017-06-15:
### Package: coreutils-stdbuf in Arm7 version repository

- expected behavior:

stdbuf should operate passively, giving no error. For example on another system, I can get the following output:

user@server:~$ stdbuf -o0 date
Wed Jun 14 22:44:00 CDT 2017
user@server:~$ 

- actual behavior: 

stdbuf reports a missing shared library, e.g.:

root@DD-WRT:~# stdbuf -o0 date
stdbuf: failed to find 'libstdbuf.so'
root@DD-WRT:~# 

### Platform:

I am running DD-WRT (Brainslayer, DD-WRT v3.0-r32170 std (06/01/17)) on a Linksys WRT1900ACS (v2 hardware). Processor is a Marvel Armada 385.

All other Entware utilities that I have installed appear to be working.

zyxmon, 2017-06-15:
Entware version of stdbuf works and does not require libstdbuf:

```
[code]
root@ws880:/tmp/home/root# opkg install coreutils-stdbuf
Installing coreutils-stdbuf (8.23-2) to root...
Downloading http://pkg.entware.net/binaries/armv7/coreutils-stdbuf_8.23-2_armv7soft.ipk
Configuring coreutils-stdbuf.
root@ws880:/tmp/home/root# opkg files coreutils-stdbuf
Package coreutils-stdbuf (8.23-2) is installed on root and has the following files:
/opt/bin/stdbuf
root@ws880:/tmp/home/root# stdbuf --help
Usage: stdbuf OPTION... COMMAND
Run COMMAND, with modified buffering operations for its standard streams.
.....
root@ws880:/tmp/home/root# ldd /opt/bin/stdbuf
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x40201000)
        libc.so.6 => /opt/lib/libc.so.6 (0x4021c000)
        /opt/lib/ld-linux.so.3 (0x4007a000)
```
Please check if you have another stdbuf binary on your system.

nfitzkee, 2017-06-15:
I only have one version installed, although my version also looks for linux-vdso.so.1 in the shared libraries list. Other than that, I get all the same results that you do, until I try to actually use stdbuf, when it returns the error.  See below.

`
root@DD-WRT:/opt/bin# opkg files coreutils-stdbuf
Package coreutils-stdbuf (8.23-2) is installed on root and has the following files:
/opt/bin/stdbuf
root@DD-WRT:/opt/bin# which stdbuf
/opt/bin/stdbuf
root@DD-WRT:/opt/bin# stdbuf --help
Usage: stdbuf OPTION... COMMAND
Run COMMAND, with modified buffering operations for its standard streams.
...
root@DD-WRT:/opt/bin# stdbuf --version
stdbuf (GNU coreutils) 8.23
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Padraig Brady.
root@DD-WRT:/opt/bin# ldd stdbuf
        linux-vdso.so.1 (0xbed2a000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0xb6f05000)
        libc.so.6 => /opt/lib/libc.so.6 (0xb6dc7000)
        /opt/lib/ld-linux.so.3 (0xb6f20000)
root@DD-WRT:/opt/bin# stdbuf -o0 date
stdbuf: failed to find 'libstdbuf.so'
`

zyxmon, 2017-06-15:
Please download http://pkg.entware.net/binaries/armv7/test/libstdbuf.so and put it in /opt/bin (not in /opt/lib !!!!). Check if it fixes the problem.
The issue is opened. Makefile needs small fix like here - https://github.com/openwrt/packages/issues/1654

nfitzkee, 2017-06-15:
This got rid of the error, but I'm not sure the behavior (actually changing the buffering behavior) is correct.  I need to get to bed, but I will troubleshoot tomorrow and see if I can find a simple test case to demonstrate whether the fix is working properly or not. Thanks!

nfitzkee, 2017-06-17:
stdbuf still seems to be problematic, even though the program does run. I cannot confirm that it's actually adjusting the buffer.

Unfortunately I still haven't worked out a "Knuth test" that clearly proves that stdbuf isn't working properly, even if it does run.  My testing is coming from an older router running a version of the original optware. stdbuf works properly there, but it doesn't work on the new install with Entware. However, the full details would be rather convoluted and difficult to test by a someone else.

That said, I'm still working on this; I've downloaded and compiled Entware-ng-3x from source, and I've been able to make my own "libstdbuf.so.  I will see if I can make any progress on my own.  Will update soon.

Finally, however, note however that /opt/bin must be first in the PATH for some reason.  See below:

```
root@DD-WRT:~# echo $PATH
/bin:/usr/bin:/sbin:/usr/sbin:/jffs/sbin:/jffs/bin:/jffs/usr/sbin:/jffs/usr/bin:/mmc/sbin:/mmc/bin:/mmc/usr/sbin:/mmc/usr/bin:/opt/sbin:/opt/bin:/opt/usr/sbin:/opt/usr/bin
root@DD-WRT:~# which stdbuf
/opt/bin/stdbuf
root@DD-WRT:~# stdbuf -o0 date
Error relocating /opt/bin/libstdbuf.so: __fprintf_chk: symbol not found
root@DD-WRT:~# export PATH="/opt/bin:$PATH"
root@DD-WRT:~# stdbuf -o0 date
Fri Jun 16 20:32:32 CDT 2017
```

zyxmon, 2017-06-17:
You can also post your findings upstream because openwrt does not place libstdbuf.so in the package. Our installation scripts put /opt/bin first in the PATH through superuser (root or admin) profile. I'll try to fix libstdbuf.so path problem later.

nfitzkee, 2017-06-18:
Agreed.  You can close this ticket.  I was able to get libstdbuf to compile/install using a very similar patch to what was included for OpenWRT. See below:

```
--- backup/Makefile	2017-06-16 18:21:09.000000000 -0500
+++ Makefile	2017-06-16 19:33:16.000000000 -0500
@@ -108,6 +108,10 @@
   define Package/$(1)/install
 	$(INSTALL_DIR) $$(1)/opt/bin
 	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/opt/bin/$(2) $$(1)/opt/bin/
+
+    ifeq ($(2),stdbuf) 
+	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/opt/lib/coreutils/libstdbuf.so $$(1)/opt/bin/ 
+    endif
   endef
 
   $$(eval $$(call BuildPackage,$(1)))
```

However, even with that fix, I was still unable to get the program to work properly.  Using the latest coreutils (8.27) also did not fix the problem. Unfortunately, I was unable to get 8.4 to compile, probably because of issues with an older version of autoconf. I know version 8.4 has the potential to work properly, as it was working in Optware (OTRW2) on my old router.

I believe this problem is related to another open ticket at OpenWRT (https://github.com/openwrt/packages/issues/1674).  That site also has an elegant way to test whether stdbuf is working. It may be a bug with the OpenWRT implementation or coreutils itself.  We'll see, but it doesn't appear to be an Entware specific problem.  As you said, I will post upstream and see what progress I can make.

Thanks for the help - stepping into github, taking a foray into cross-compiling, and hacking around in OpenWRT has certainly been educational.

closed 2017-06-18 by nfitzkee