Entware-for-kernel-3x/Entware-ng-3x #21: not working either, i fixed the install, it should have its own wget but i get an error

p3x-robot, 2017-06-15:
```text
root@router:/opt# opkg update
Downloading http://entware-3x.zyxmon.org/binaries/armv7/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/packages
Collected errors:
 * parse_from_stream_nomalloc: Missing new line character at end of file
```

nfitzkee, 2017-06-17:
Hard to tell what's actually happening since you didn't post your script or details. However, I had to pretty much run the entire install script manually, and then I needed to grab several files and transfer them by ssh before entware would work.   

More precisely, I used SSH/SFTP (curl would work, too) to transfer the libraries, /opt/etc/opkg.conf, opkg, and Packages.gz manually. You have the URL for these files in the install script. Presumably you've already gotten these. Packages.gz (and the files below) are available at http://entware-3x.zyxmon.org/binaries/armv7/. Once the file is transferred to your router, just copy Packages.gz to:

```
/opt/var/opkg-lists/packages
```

At this point, opkg should be able to recognize and hash all package files, but because of the faulty busybox-wget, it still won't be able to get anything.

The trick now is to install wget using opkg, but without using the faulty wget to download any files.  To do that, I needed to get its dependencies  and install them manually.  I determined the dependencies by trial-and-error, but the list I ended up needing is below. The URL for these files is the same: http://entware-3x.zyxmon.org/binaries/armv7/.  I just downloaded them via the web and transferred via my trusty GUI SFTP client, but there are lots of other ways to do this with scripting, curl, etc.

```
libc_2.25-8_armv7-3x.ipk          
librt_2.25-8_armv7-3x.ipk
libgcc_6.3.0-8_armv7-3x.ipk       
libssp_6.3.0-8_armv7-3x.ipk
libopenssl_1.0.2k-1_armv7-3x.ipk  
wget_1.18-2_armv7-3x.ipk
libpcre_8.40-2_armv7-3x.ipk
zlib_1.2.11-1_armv7-3x.ipk
libpthread_2.25-8_armv7-3x.ipk
```

Once downloaded and transferred to your router, you can install these packages at the command line by simply typing "opkg install <list of files>"  If you've done it right, there will be no need to download additional files using wget.  If additional files are needed, simply download those manually, transfer via ssh/sftp to your router, and add the additional files to command line.

Once the files are installed, you will have a non-busybox version of wget which can be used to obtain the other packages.  Moreover, your opkg database will be more or less intact. However, you will still need to update your path so that opkg favors the correct wget binary; in my case (DD-WRT), that was

```
export PATH="/opt/bin:$PATH"
```

You will need to update your startup scripts to execute this statement at every login; alternatively, you can simply type it when you log in.

Once that's done, you should be good, but don't forget to run the any parts of the of the install script you didn't run manually!

p3x-robot, 2017-06-17:
How !

How are you?

sorry to ask, i know you are busy  yourself.
do you think?

 i am using the latest   
`ftp://ftp.dd-wrt.com/betas/2017/06-01-2017-r32170/linksys-wrt1900acsv2/`

Do you think it is too new! It is kernel 4.9?
It works and after 20  minutes later stops working, i used the arvm7 
I downloaded https://www.busybox.net/downloads/binaries/1.21.1/busybox-armv7l, and linked wget there and then wget works.
But then after a while i get  ```parse_from_stream_nomalloc: Missing new line character at end of file```

I got back to factory Linksys 1900ACS, but I would love to use DDWRT. (Cant use HE-IPV6 statis address)

Should I use LEDE? I love DDWRT , so easy.

SO how do I install the packages from http://entware-3x.zyxmon.org/binaries/armv7/ if the opkg gives me to weird errors .

Which router and firmware do you use?

Thanks when you get a chance!

zyxmon, 2017-06-17:
Once you install entware-3x - you do not need to install wget. Entware-3x installs busybox with working wget. Just put /opt/bin firts in your PATH.
PS To install Entware find a working static build of busybox somewhere in the internet and put temporary wget first in your PATH.

closed 2017-06-17 by zyxmon

p3x-robot, 2017-06-18:
YOU ARE A GOD! I FIXED IT! I DID WHAT YOU SAID!
YOU ARE SO COOL!
THANKS SO MUCH!
YOU ARE THE MAIN MAIN SUPER POWER GOOD!