Entware-for-kernel-3x/Entware-ng-3x #26: busybox 1.25.1: remove: can't open 'share/busybox_applets.lst': No such file or directory

ProBackup-nl, 2017-06-22:
### Package: busybox (1.25.1)
- expected behavior: no error on removal
- actual behavior: error on remove

```
# opkg remove busybox --force-removal-of-essential-packages
Removing essential package busybox under your coercion.
	If your system breaks, you get to keep both pieces
Removing package busybox from root...
cat: can't open 'share/busybox_applets.lst': No such file or directory
# updatedb && locate /busybox_applets.lst
<no output>
```
### Platform: Asus RT-N56U padavan 099, 3.4.113

zyxmon, 2017-06-22:
This is normal. Some packages from entware-3x use busybox from Entware-3x. We do not recommend to remove it.
If you prefer some bysybox applet from your firmware, just make a link from /opt/[s]bin/<applet name> to busybox from firmware.

closed 2017-06-22 by zyxmon

ProBackup-nl, 2017-06-22:
On a clean entware installation `opkg list-installed | grep busybox` doesn't output any result, does it?

zyxmon, 2017-06-22:
This is a almost clean entware-3x installtion on Padavans firmware (first command is to show it)

```
/opt/home/admin # ls /sbin/*.sh
/sbin/automount.sh    /sbin/hddtune.sh      /sbin/reset_ss.sh
/sbin/dev_init.sh     /sbin/mtd_storage.sh  /sbin/spindown.sh
/opt/home/admin # opkg list-installed
busybox - 1.25.1-2
entware-opt - 1.0-4
findutils - 4.6.0-1
glib2 - 2.50.2-2
iptables - 1.4.21-2a
libattr - 20160302-1
libblkid - 2.28-1b
libc - 2.25-8
libffi - 3.2.1-2
libgcc - 6.3.0-8
libiconv-full - 1.11.1-3
libintl-full - 0.19.8.1-1
libmount - 2.28-1b
libopenssl - 1.0.2k-1
libpthread - 2.25-8
librt - 2.25-8
libslang2 - 2.3.1-1
libssh2 - 1.7.0-1
libssp - 6.3.0-8
libstdcpp - 6.3.0-8
libuuid - 2.28-1b
libxml2 - 2.9.4-1
locales - 2.25-8
mc - 4.8.18-1a
openssh-keygen - 7.4p1-1
openssh-server - 7.4p1-1
openssl-util - 1.0.2k-1
smartmontools - 6.5-1
terminfo - 6.0-1c
whereis - 2.28-1b
zlib - 1.2.11-1
zoneinfo-asia - 2016j-1
zoneinfo-europe - 2016j-1
```

Entware installation and usage can be discussed somewhere else, not here. I can suggest http://forums.zyxmon.org/index.php as a place for such discussions. You can suggest another forum. Please post your steps and entware installation logs. My guess - you have installed entware-ng, not entware-3x.

ProBackup-nl, 2017-06-22:
I as far as I know, I have installed **Entware-ng-3x** and am still in the understanding that this repo here is Entware-ng-3x, not entware-ng neither entware-3x.

zyxmon, 2017-06-22:
@ProBackup-nl: FYI Entware-3x installation is rather tricky on Padavan firmware. It is described on the forum in Russian (the link is in my post). Please use forums for further discussions.

ProBackup-nl, 2017-06-22:
@zyxmon Thanks for the forum link, I don't understand any Russian. 

Where can I find more information regarding the trick for Entware-3x on Padavan firmware (in English, German, or Dutch).

zyxmon, 2017-06-22:
https://translate.google.com/translate?sl=ru&tl=en&js=y&prev=_t&hl=ru&ie=UTF-8&u=http%3A%2F%2Fforums.zyxmon.org%2Fviewtopic.php%3Fp%3D15556%23p15556&edit-text=

ProBackup-nl, 2017-06-22:
Thanks for the google translate link.

> These requirements are certainly met by firmware Padavana,

What I can remember is that I did an [standard _std install](http://entware-3x.zyxmon.org/binaries/mipsel/installer/install_std.sh) to maintain a single user base.

I double checked that the padavan firmware its kernel is 3.4.112 or greater (3.4.113).

After scanning thru the 13 pages of the translated forum post, I still haven't found the _tricky_ part for padavan firmware in combination with entware-ng-3x.

zyxmon, 2017-06-22:
The translation point to the post with instructions. Press `+` to expand them for Padavan firmware.

ProBackup-nl, 2017-06-22:
> In Padavan's firmware there is a built-in script that installs `Entware-ng`. We need it not to work to install `Entware-3x` manually.
> Before installation, make sure that the "_USB Applications> General Settings_" tab on the web tab of the drop-down list "_Allow Optware to Start?_" "_No_" was selected. Prepare the media according to [the instructions formatting the USB disk drive on the router with Padavan firmware](https://bitbucket.org/padavan/rt-n56u/wiki/EN/HowToConfigureEntware#markdown-header-using-the-router-to-format-the-disk-drive) until (not including) section "_Using Entware_".

> After the media with the correct file system (ext2/ext3/ext4) is connected to the router and mounted, create the `/opt/bin` folder and the `/opt/bin/opkg` file on it. If, as in the article above, the label of the sda1 section is "_Main_":
> 
> `# mkdir -p /media/Main/opt/bin;touch /media/Main/opt/bin/opkg`
>
> This will block the installation of `Entware-ng`. After that, "_enable Entware_" in the web interface ("USB Applications> General Settings> Enable Optware? - Entware"). We reboot the router, check (`mount` command), that it is mounted in `/opt`.
> `/opt/home/admin # mount | grep /opt`
> `/dev/sda1 on /opt type ext4 (rw,noatime,data=ordered)`
> After that, delete the file created by the firmware scripts:
> `# rm /opt/etc/profile`
>
> This is recommended for unification with Entware-3x installations on other devices. The action is not mandatory.
> We can install Entware. For a standard installation:
> `# wget -O - http://entware-3x.zyxmon.org/binaries/mipsel/installer/install_std.sh | /bin/sh`
> To install the recommended alternative:
> `# wget -O - http://entware-3x.zyxmon.org/binaries/mipsel/installer/install_alt.sh | /bin/sh`
> In case you used an alternative, it is recommended to install ssh server `openssh` (read the brief instructions below in this thread). Restart the router, then use port 2222 for ssh, `root` login, password `12345`. You can change the password after login, add additional users.
> Set the time zone for your region (see the FAQ in the topic). For Amsterdam it is done this way:
> `# ln -sf /opt/share/zoneinfo/Europe/Amsterdam /opt/etc/localtime`
> The installation is complete.

ProBackup-nl, 2017-06-22:
After reading the Google translated instructions, and:
```
ls -la /opt/bin/opkg
-rwxrwxrwx    1 admin   root        471924 Apr 28  2016 /opt/bin/opkg
```
It looks that Padavan firmware has overwritten the `138836 bytes` and `20-Feb-2017 05:26` dated entware-3x version of `opkg`.

Now I do understand _tricky_. (let me give some thoughts on how to clean up this mess)

ProBackup-nl, 2017-06-22:
@zyxmon Pleasy copy the quoted text from https://github.com/Entware-for-kernel-3x/Entware-ng-3x/issues/26#issuecomment-310390875 to a [new wiki page here](https://github.com/Entware-for-kernel-3x/Entware-ng-3x/wiki/General-installation-notes) ("_Install on Padavan_")

ProBackup-nl, 2018-02-06:
For `crontab` it seems best to use Padavan firmware its crontab, thus:
````
rm /opt/bin/crontab
````