Entware-for-kernel-3x/Entware-ng-3x #31: QNAP x86 support request

joahallin, 2017-07-29:
### Package: Entware-ng-3x

Please add support for x86 (32-bit)

### Platform:

QNAP TS-259 Pro+, Kernel 3.4.6 i686

zyxmon, 2017-07-29:
Please use Entware-ng in case your cpu is 32 bit only.

closed 2017-07-29 by zyxmon

joahallin, 2017-07-29:
It seems Entware-ng is no longer compatible with the latest QTS 4.2.6 (2017/06/28).  I've tried installing Entware-ng_0.97.qpkg and the package ends up in /share/MDO_DATA/.qpkg/Entware-ng but the installer fails without any specific error message and there's no Entware content in /etc/config/qpkg.conf. I believe I used to have Entware-ng installed but then some recent firmware update seems to have broken the installation and again, reinstallation is not successful. That's why I wanted to use Entware-ng-3x in the hope that it's compatible with the latest firmware. Any advice on how to proceed?

zyxmon, 2017-07-29:
Please post installation log on qnap forum - https://forum.qnap.com/viewforum.php?f=351
Download Entware-3x qpkg and run `sh <path_to_qpkg>` in console.
My guess that your cpu and kernel support x64 and x64 repo is used for Entware-ng and Entware-3x. Just run `uname -m` to check it.