Entware-for-kernel-3x/Entware-ng-3x #32: Net::SSLeay install

joahallin, 2017-08-06:
### Package: Entware-ng-3x

This might not be the right forum but I've tried installing the perl modules  Net::SSLeay and IO::Socket::SSL using the method outlined in the Entware-ng wiki. This doesn't quite work. 
There is a warning message when installing Net::SSLeay:
Warning (mostly harmless): No library found for -lssl
Warning (mostly harmless): No library found for -lz

However, the installation seems to go through and testing afterwards seems fine:
 perl -e "use Net::SSLeay"

Proceeding then by installing IO::Socket::SSL seems to go through but when testing it fails like this:
 perl -e "use IO::Socket::SSL"
perl: symbol lookup error: /opt/lib/perl5/5.24/auto/Net/SSLeay/SSLeay.so: undefined symbol: SSL_library_init

Is the openssl library missing? I don't seem to have that library, only libgnutls-openssl (which might be unrelated). This might not be the issue though?

Please assist.

### Platform:
QNAP 215-Pro, Linux server 4.2.8 #1 SMP Thu Jul 27 01:09:44 CST 2017 x86_64 GNU/Linux, QTS 4.3.3.0262 (2017/07/27), Entware-3x_0.99std

zyxmon, 2017-08-06:
1. It is only a warning
2. All libraries needed for compilation should be installed (ordinary version, no dev versions in enware). Please check that you have installed libopenssl and zlib packages.
3. You may need to patch installation files in some cases.

closed 2017-08-06 by zyxmon

joahallin, 2017-08-06:
Both libopenssl and zlib are installed according to opkg. Why do I get the warning message -lssl and -lz then? Some path problem? I downloaded the latest versions of Net::SSLeay and IO::Socket::SSL from cpan.
Any other hints that you can provide?

zyxmon, 2017-08-06:
Post https://forum.qnap.com/viewtopic.php?p=606772#p606772 may help

>> For those trying to install Net::SSLeay this CPAN line worked for me
>> o conf makepl_arg 'LIBS="-L/opt/lib -lcrypto -lssl"'

joahallin, 2017-08-08:
Thanks!
That did the trick for the 64-bit perl version. Unfortunately it failed
when trying to do the same for the 32-bit perl which is part of the
LogitechMediaServer. This is not really surprising since gcc is 64-bit,
openssh is 64-bit. I guess getting that to work I would have to replace
that perl version, and all of the included modules with 64-bit perl which
seems to be challening...