Entware-for-kernel-3x/Entware-ng-3x #38: make V=99 failed

debiansid, 2017-09-24:
I was trying to compile who toolchain locally on debian 9.1. But stop at 
````
install: failed to access '/home/storage/Entware-ng-3x/staging_dir/toolchain-arm_cortex-a9_gcc-_-_eabi/bin/arm-openwrt-linux-gdb': Too many levels of symbolic links             
Makefile:68: recipe for target '/home/storage/Entware-ng-3x/staging_dir/toolchain-arm_cortex-a9_gcc-_-_eabi/stamp/.gdb_installed' failed                                         
make[3]: *** [/home/storage/Entware-ng-3x/staging_dir/toolchain-arm_cortex-a9_gcc-_-_eabi/stamp/.gdb_installed] Error 1
make[3]: Leaving directory '/home/storage/Entware-ng-3x/toolchain/gdb'
toolchain/Makefile:96: recipe for target 'toolchain/gdb/compile' failed
make[2]: *** [toolchain/gdb/compile] Error 2
make[2]: Leaving directory '/home/storage/Entware-ng-3x'
toolchain/Makefile:94: recipe for target '/home/storage/Entware-ng-3x/staging_dir/toolchain-arm_cortex-a9_gcc-_-_eabi/stamp/.toolchain_compile' failed
make[1]: *** [/home/storage/Entware-ng-3x/staging_dir/toolchain-arm_cortex-a9_gcc-_-_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/storage/Entware-ng-3x'
/home/storage/Entware-ng-3x/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
````

What’s the catch?

zyxmon, 2017-09-24:
We do know there are problems with debian 9.
Please use Debian 8 until we fix buildroot for debian 9 .

PS It looks like you have not copied appropriate config file.

debiansid, 2017-09-24:
I re-run make menuconfig ; chose to not compile gdb, it keeps going now.
let's see where it goes.

debiansid, 2017-09-24:
ok, I made it work.here is the problem

1.staging_dir/host/usign is missing

2.root dir/key-build is missing

zyxmon, 2017-09-30:
To use debian 9 one should uninstall libssl-dev (`sudo apt remove libssl-dev`) and install libssl1.0-dev (`sudo apt install libssl1.0-dev`).
Please use appropriate config file from configs dir to build packages.

closed 2017-09-30 by zyxmon