Entware-for-kernel-3x/Entware-ng-3x #42: parse_from_stream_nomalloc: Excessively long line at 11

liquidwax, 2017-09-28:
### Package: n.a.
### Platform: ac56u & ac68u both running both running DD-WRT v3.0-r32170M kongac firmware.

opkg upgrade ended with the usual errors on conf files conflicts.
When I tried to do it again, I get the 

Here's the tail from the output when opkg was run with -V4:
pkg_vec_insert_merge: Adding new pkg=libunrar version=5.4.5 arch=armv7-3x.
hash_table: pkg-hash, 12288 bytes
 	n_buckets=1024, n_elements=942, n_collisions=326
 	max_bucket_len=4, n_used_buckets=616, ave_bucket_len=1.53
 	n_hits=4840, n_misses=942
hash_table: file-hash, 12288 bytes
 	n_buckets=1024, n_elements=0, n_collisions=0
 	max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
 	n_hits=0, n_misses=0
hash_table: obs-file-hash, 768 bytes 
 	n_buckets=64, n_elements=0, n_collisions=0
 	max_bucket_len=0, n_used_buckets=0, ave_bucket_len=0.00
 	n_hits=0, n_misses=0
Collected errors:
 * parse_from_stream_nomalloc: Excessively long line at 11. Corrupt file?
I'm not sure if it's just me but I got the errors when done on another router.

zyxmon, 2017-09-28:
I think your /opt/lib/opkg/status file is corrupted now.
1) Send it to me <my nik>@gmail.com (or both files)
2) This is a text file. Try to find out what is wrong and fix it with text editor.
This can be caused by some incorrect updated package or by a bug in opkg.
You can get the list of installed packages manually - each installed has corresponding info file in /opt/lib/opkg/info filder. This can help to restore status file.

liquidwax, 2017-09-30:
Sorry for not replying earlier.
I've tried looking at the /opt/lib/opkg/status file but couldn't find any abnormalities with my abilities.
I've sent you the status files but you may ignore them since I've already done a fresh install and succeeded without errors.
Thanks.

closed 2017-09-30 by liquidwax