Entware-for-kernel-3x/Entware-ng-3x #44: mkvtoolnix looking for previous version boost libraries

shawnhicks, 2017-10-02:
### Package: mkvtoolnix - 12.0.0-1
For existing package:

- expected behavior: ...
finds 1.65.0-1 boost libraries
- actual behavior: ...
[~] # ldd `which mkvmerge`
        linux-vdso.so.1 (0x00007ffd9f2fd000)
        libstdc++.so.6 => /opt/lib/libstdc++.so.6 (0x00007f8ae6bf3000)
        libintl.so.8 => /opt/lib/libintl.so.8 (0x00007f8ae69e9000)
        libmagic.so.1 => /opt/lib/libmagic.so.1 (0x00007f8ae67ca000)
        libz.so.1 => /opt/lib/libz.so.1 (0x00007f8ae65b4000)
        libmatroska.so.6 => /opt/lib/libmatroska.so.6 (0x00007f8ae62ed000)
        libebml.so.4 => /opt/lib/libebml.so.4 (0x00007f8ae60c8000)
        libiconv.so.2 => /opt/lib/libiconv.so.2 (0x00007f8ae5de3000)
        libboost_regex.so.1.64.0 => not found
        libboost_filesystem.so.1.64.0 => not found
        libboost_system.so.1.64.0 => not found
        libFLAC.so.8 => /opt/lib/libFLAC.so.8 (0x00007f8ae5b72000)
        libogg.so.0 => /opt/lib/libogg.so.0 (0x00007f8ae596b000)
        libvorbis.so.0 => /opt/lib/libvorbis.so.0 (0x00007f8ae5740000)
        libm.so.6 => /opt/lib/libm.so.6 (0x00007f8ae542f000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x00007f8ae5218000)
        libc.so.6 => /opt/lib/libc.so.6 (0x00007f8ae4e7d000)
        /opt/lib/ld-linux-x86-64.so.2 (0x00007f8ae6f72000)
        librt.so.1 => /opt/lib/librt.so.1 (0x00007f8ae4c75000)
        libpthread.so.0 => /opt/lib/libpthread.so.0 (0x00007f8ae4a57000)

For new package to be added:

- make sure there's no similar package already added,
- describe killer features of new package,
- make sure it's open source and not abandoned (at least for last 5 years or so),
- please, use native compilation for additional PHP, Perl, Ruby, Python modules for non-MIPS platforms.

### Platform:
qnap ts870 firmware 4.3.3.0299 - Entware-3x 0.98std

- please, tell us what feed and\or devices are you using.
- be ready to check issue-related  changes or it will be closed after 30 days with "no response" status.

shawnhicks, 2017-10-02:
mkvtoolnix release 16.0.0 is available, maybe an update is in order?

zyxmon, 2017-10-02:
Dublicate is with Entware-ng.

closed 2017-10-02 by zyxmon