Entware-for-kernel-3x/Entware-ng-3x #45: dnsmasq < 2.78 (Security vulnerabilities)

jamesmacwhite, 2017-10-02:
### Package: dnsmasq

Due to several security issues/bugs found with dnsmasq versions < 2.78 recently, is it possible sync the recent changes to dnsmasq from LEDE into Entware?

I'm aware, that a sync happens every few months or so, but due to the vulnerability/security nature of this update, is it possible to push these changes up?

https://github.com/lede-project/source/commit/67ac017fefe8552b4e864d0494e892efe4fb9710
https://security.googleblog.com/2017/10/behind-masq-yet-more-dns-and-dhcp.html

Various CVE's have been published.

zyxmon, 2017-10-03:
Done. arm & intel feeds updated. mips & mipsel will be updated soon.

closed 2017-10-03 by zyxmon

jamesmacwhite, 2017-10-03:
@zyxmon Many thanks for this! Much appreciated!