Entware-for-kernel-3x/Entware-ng-3x #46: rrdtool-1.2.30-3a - deteriorated package

zd59, 2017-10-04:
### Package: <rrdtool-1.2.30-3a>

Package issued: 25-Sep-2017 04:57
For existing package:

- expected behavior: full package
- actual behavior: package deteriorated, unusable

This package is severely deteriorated and so unusable!
Entware opkg install rrdtool also installed all dependencies, so above is reality.
After adding several missing pm modules (RRDs.pm, DynaLoader.pm ..) I crashed to wall below:

Wed Oct  4 10:59:37 2017 - WARNING: unable to load module 'system'. Can't locate loadable object for module RRDs in @INC (@INC contains: /tmp/mnt/entware/entware/home/monitorix/lib /usr/lib/monitorix /opt/lib/perl5/5.24 .) at /tmp/mnt/entware/entware/home/monitorix/lib/system.pm line 26.
Compilation failed in require at /tmp/mnt/entware/entware/home/monitorix/lib/system.pm line 26.

As error above suggests, there is at least one missing object module RRDs.so (highly possible more missing).

Please rectify this package as soon as possible with adding ALL missng pm modules and dependent library objects (*.so).

Wish to add Monitorix [http://www.monitorix.org/](url), which is rrdtool dependent, but it is impossible for the above reason.

### Platform:

- the device used was Netgear R7500 V1 Router, Firmware Version (Voxel) V1.0.1.13SF + Entware-ng-3x from 25-Sep-2017

zyxmon, 2017-10-04:
rrdtool package does not have perl modules as dependencies.
Previous version of Entware-3x does not have `RRDs.so` in all packages.

Please give the exact instructions to reproduce the bug with the fresh entware-3x install.

zd59, 2017-10-04:
Thank you for a fast response!

Expecting Monitorix installation problems, I written every step I done.
So I attahed two files:
1. R7500_install_Monitorix.txt - Monitorix install steps
2. opkg_installed_20171004.txt - date & list of all installed Entware
packages.

Please advice me with further instructions to correctly install all needed
rrdtool packages, so Monitorix will work.

Big thanks in advance!

Best regards,

Zdenko Dolar

zd59, 2017-10-04:
Thank you for a fast response!

Expecting Monitorix installation problems, I written every step I done.
So I attahed two files:
1. R7500_install_Monitorix.txt - Monitorix install steps
2. opkg_installed_20171004.txt - date & list of all installed Entware packages.

Please advice me with further instructions to correctly install all needed rrdtool packages, so Monitorix will work.

Big thanks in advance!

Best regards,

Zdenko Dolar

[opkg_installed_20171004.txt](https://github.com/Entware-for-kernel-3x/Entware-ng-3x/files/1355796/opkg_installed_20171004.txt)
[R7500_install_Monitorix.txt](https://github.com/Entware-for-kernel-3x/Entware-ng-3x/files/1355797/R7500_install_Monitorix.txt)

zyxmon, 2017-10-04:
Please read https://github.com/Entware-ng/Entware-ng/wiki/Self-installation-of-perl-modules
(take into account entware-ng and entware-3x differences).
The problem you have does not look like entware bug. Closing.

closed 2017-10-04 by zyxmon

zyxmon, 2017-10-05:
>> What do you mean by last sentence above - .. in all packages?

If you (fresh) install all 2000+ packages, you will not find RRDs.so file in /opt dir.

>>  but configure complains "gcc can not compile"

Have you installed gcc and headers as wiki suggests? If gcc was installed - have you examined config.log file?

zd59, 2017-10-06:
Hi Again!

Followed your instructions to the letter:
[https://github.com/Entware-ng/Entware-ng/wiki/Self-installation-of-perl-modules](url)
and from there:
[https://github.com/Entware-ng/Entware-ng/wiki/Using-gcc-%28native-compilation%29](url)
Installed headers are from [http://pkg.entware.net/binaries/armv7/include/include.tar.gz](url)

rrdtool-1.2.30 sources, error at:
./configure --prefix=/opt

I included text file with error and following my system configuration and config.log generated.
[rrdtool_config_error.txt](https://github.com/Entware-for-kernel-3x/Entware-ng-3x/files/1362393/rrdtool_config_error.txt)
[config.log](https://github.com/Entware-for-kernel-3x/Entware-ng-3x/files/1362382/config.log)

zyxmon, 2017-10-06:
1. You have used the instructions for Entware-ng without taking into account, that you use Entware-3x (by voxel).
2. The instructions are for perl modules, not for rrdtool.
3. We do not provide support for native compilation, sorry. We do not recommend it, you can use it at your own risk.

zd59, 2017-10-06:
without taking into account, that you use Entware-3x - Yes I am aware of that. 
Can you please point out Entware-3x variant? Can not find it myself.