Entware-for-kernel-3x/Entware-ng-3x #47: par2cmdline - not multi-threaded?

teracow, 2017-10-04:
### Package: par2cmdline

For existing package:

- expected behavior: ...
to use parameter '-t' for multi-thread processing
- actual behavior: ...
this parameter is not present in packaged version

### Platform:
x86/64

Hi zyxmon, I've just noticed that the version of par2cmdline (0.7.4-1) available here is missing the multi-threading option. Would it be possible to include the multi-thread version instead? 

https://github.com/Parchive/par2cmdline

Thank you. :)

zyxmon, 2017-10-04:
Entware buildroot is based on lede/openwrt buildroot, that does not have openmp support needed for par2cmdline-mt,

closed 2017-10-04 by zyxmon

teracow, 2017-10-04:
Ah, no worries. Thank you.