Entware-for-kernel-3x/Entware-ng-3x #54: Two issues with git package

gmerrall, 2018-01-01:
### Package: git

_Issue 1._  The template directory is not populated. Normally when you run "git init --bare" a number of files are copied over including the contents of the 'template' directory which typically contains sample scripts. 

Perform the following steps on a regular box with git installed
```
$ strings `which git` | grep /templates
share/git-core/templates
$ ls /usr/share/git-core/templates/
branches	description	hooks		info
```

If this is done with a Entware-ng-3x install
```
$ strings `which git` | grep /templates
/opt/share/git-core/templates
$ ls /opt/share/git-core/templates
<empty directory>
````

Expected result is the template files should be present

_Issues 2._  No HTTPS support. This is likely due to it not being compiled with curl-devel support.

Normal box:
```
$ git clone https://github.com/greg0ire/git_template
Cloning into 'git_template'...
```

If this is done with a Entware-ng-3x install
```
$ git clone https://github.com/greg0ire/git_template
Cloning into 'git_template'...
fatal: Unable to find remote helper for 'https'
```

Box: QNAP 
Linux NASDXXXXXX 3.4.6 #1 Wed Dec 13 09:02:49 CST 2017 armv5tel GNU/Linux

zyxmon, 2018-01-01:
1. templates are missing upstream. Probably (it is common for openwrt packages) space is preserved this way. Please report upstream if you think it is a bug.
2. Please install `opkg install git-http` for https support!

closed 2018-01-01 by zyxmon

gmerrall, 2018-01-03:
I think you may want to resolve this issue Entware-ng-3x.  I suspect strongly that the reason they were removed upstream is that that they're only useful if you're using git as a server. In openwrt it would exclusively be used as a client so these files aren't required and they would have been removed to save space. 
I'll file an issue upstream but suspect it will be closed for the reasons I've outlined