Entware-for-kernel-3x/Entware-ng-3x #55: Package Request: brlaser

ian5142, 2018-01-01:
Package: brlaser
https://github.com/pdewacht/brlaser

Driver for (some) Brother laser printers

Most Brother printers support a standard printer language such as PCL or PostScript, but not all do. If you have a monochrome Brother laser printer (or multi-function device) and the other open source drivers don't work, this one might help.

It is known to support these printers:
    Brother DCP-1510 series
    Brother DCP-7030
    Brother DCP-7055
    Brother DCP-7055W
    Brother DCP-7065DN
    Brother MFC-7360N
   + other Brother DCP/MFC series printers (not tested but likely)

Platform: MIPS based TP-Link Archer C7 v2, running dd-wrt.

The-BB, 2018-01-02:
Can you try [this package](http://entware-3x.zyxmon.org/binaries/mips/test/brlaser_3-1_mips-3x.ipk)?
Have a nice day!

ian5142, 2018-01-03:
It installed but I haven't tested it yet as my cups installation just went out of whack. I will get back to you on that.

ian5142, 2018-01-03:
It should be noted that brlaser does have a hidden requirement for ghostscript. Also, I can't get the drivers to show up in cups. Cups does not list any new Brother printer models that gutenprint does not already support.

zyxmon, 2018-01-03:
We do not have brother printers to check brlaser. If we find how to fix the packages - we will make another test package.

ian5142, 2018-01-03:
I have several Brother printers that I can use to test the package but I don't know exactly how to do that. Can you explain exactly how you want me to test the package? It installed fine, I just can't figure out what to do next.

zyxmon, 2018-01-03:
I do not use cups. May be this is the starting point - https://wiki.openwrt.org/doc/howto/cups.server

ian5142, 2018-01-04:
Where does the package place files (or install to)?

ian5142, 2018-01-04:
What folders under /opt (my root directory)?

zyxmon, 2018-01-04:
Use `opkg files <package_name>` to display the info on installed package. For example `opkg files cups`.

ian5142, 2018-01-04:
It did install. I just thought it was going to create PPD files. It doesn't. That means I also require ppdc (PPD compiler for cups): https://www.cups.org/doc/postscript-driver.html#FIGURE_1
The package you sent created a brlaser.drv file, which can then produce several PPD files that cups can use to define printers.

zyxmon, 2018-01-05:
The package @The-BB has built just creates everything as designed by the author.
`cups-ppdc` package contains ppdc. Have you managed to create ppd on your router and use Brother printers?

ian5142, 2018-01-05:
I installed cups-ppdc. I am getting an error when I run ppdc: "ppdc: Unable to find include file "<font.defs>" on line 20 of brlaser.drv." 

I think it can't find this file: https://opensource.apple.com/source/cups/cups-327/cups/data/font.defs.auto.html

zyxmon, 2018-01-05:
cups is splited into 10 smaller packages in Openwrt and Entware (`opkg list cups*`), but the file `font.defs` is not included in packages. I have asked @The-BB to make full cups.tar.gz with all files. I do not build mips and mipsel repo. The tar.gz will be in http://entware-3x.zyxmon.org/binaries/mips/test/ soon. You will find `font.def` in /opt/share/cups.... there.

ian5142, 2018-01-05:
I got it to produce the PPD files. The full command I needed, from the /opt/share/cups/drv/ directory, was:  ppdc -I /opt/share/cups -d /opt/etc/cups/ppd -m brlaser.drv

I am still testing to see if the driver works.