Entware-for-kernel-3x/Entware-ng-3x #56: imagemagick

eliz82, 2018-01-05:
hello. i have opened two issues regarding imagemagick on openwrt git.
they do not have a maintainer for that and suggested to open the issues on entware.
https://github.com/openwrt/packages/issues/5350
https://github.com/openwrt/packages/issues/5349

zyxmon, 2018-01-05:
Only Voxel can build ipk package for you. 
Wait for Voxel to update ImageMagic to 7.0.7 (as in Entware-3x). Try to unpack this file - http://entware-3x.zyxmon.org/binaries/armv7/test/xml.tar.gz it has missing xml files. If this fixes the problem, we will update ipk with next build.
PS You can temporary uninstall Voxel variant and install Entware-3x to speed up things.

eliz82, 2018-01-05:
1) copying that files to /opt/etc fixed the problem, no error anymore when making a simple convert. if i delete the directory the error reaper ... so it seems it's ok.

2) what about the second issue? supporting more files formats like BPG and WEBP ?

`magick -version`
will return this:

Version: ImageMagick 7.0.7-4 Q16 arm 2017-09-28 http://www.imagemagick.org
Copyright: © 1999-2017 ImageMagick Studio LLC
License: http://www.imagemagick.org/script/license.php
Features: Cipher DPC HDRI Modules
Delegates (built-in): jng jpeg ltdl png tiff zlib

zyxmon, 2018-01-06:
Thank for fast testing. We have upgraded ImageMagick to 7.0.7-19 and added missing xml files to ipk via https://github.com/Entware-for-kernel-3x/entware-oldpackages-ports-3x/commit/1e4b4793ac61c413399a84661d4a0f016b713d0b
As for new formats. BPG is supported with external encoder/decoder binaries that are not part of ImageMagick. To add WEBP support we need to add дшицуиз library from https://developers.google.com/speed/webp/
We need more request for new formats to consider adding them to imagemagick,
I am keeping this issue open for such requests,

eliz82, 2018-01-06:
thanks for the response. 
Also for the future you can consider HEIF format, it's similar with BPG (using image/s encoded with HVEC). But HEIF is ISO standardized and it was adopted in iOS 11 as a JPEG replacer. I don't use iPhone but i have readed they use this format when tacking camera pictures.
It seems they want to implement this https://github.com/ImageMagick/ImageMagick/issues/507

closed 2018-02-06 by zyxmon