Entware-for-kernel-3x/Entware-ng-3x #57: The output of uname changed for me

1951FDG, 2018-01-14:
### Entware-ng-3x

I recently, just now, installed Entware-ng-3x on my RT-AC86U router, to find out that the output of `uname -o` is no longer the same! It now returns GNU/Linux instead of ASUSWRT-Merlin, how has this happened or better yet, how I can fix this?
- expected behavior: Linux RT-AC86U-E5F0 4.1.27 #2 SMP PREEMPT Sat Dec 2 01:19:36 EST 2017 aarch64 ASUSWRT-Merlin

- actual behavior: Linux RT-AC86U-E5F0 4.1.27 #2 SMP PREEMPT Sat Dec 2 01:19:36 EST 2017 aarch64 GNU/Linux

### Platform:

- RT-AC86U on ASUSWRT-Merlin running working Entware-ng-3x

zyxmon, 2018-01-15:
Entware-3x installs its own busybox copy. The PATH variable has /opt/bin and /opt/sbin before /bin and /sbin. New uname version applet comes first now. You can check it by running
`/bin/uanme -a` and `/opt/bin/uname -a`.
You can simply remove `rm /opt/bin/uname` if you do not need it.

closed 2018-01-15 by zyxmon

1951FDG, 2018-01-15:
Hi, thanks for the valuable info! I guess that's what you meant on the readme [page](https://github.com/Entware-for-kernel-3x/Entware-ng-3x) , `busybox from Entware is forced instolled`