Entware-for-kernel-3x/Entware-ng-3x #59: What are the benefits of Entware-ng-3x over Entware-ng?

ProBackup-nl, 2018-02-06:
# Entware-3x benefits?
Platform: Padavan RT-N56U

One of the benefits of Entware-3x over Entware-ng is the ability to use customs list of usernames and groups that differs from the router/NAS firmware its `/etc/passwd` and `/etc/group` and those are stored at `/opt`.

What are the other benefits of Entware-3x over Entware-ng?

zyxmon, 2018-02-06:
The feature you quoted is usefull for devices where /etc/passwd is readonly, for android devices.
Here is what I've posted on asuswrt forum
"Entware-3x is new version of Entware optimized for 3.x and 4.x linux kernels. It is based on Entware-ng but can be used only on firmwares with 3.x and 4.x linux kernels. Entware-3x has more packages compared to Entware-ng because some packages cannot be compiled for old kernels, some Entware-3x packages have higher versions for the same reason. 

Entware-3x is built for mips, mipsel, armv5, armv7, aarch64, x64 platforms."

Some Entware-3x binaries are faster than Entware-ng.
Compilnig modern packages for Entware-ng has become a big headache because of 2.6 kernel.

ProBackup-nl, 2018-02-07:
Found [the July 26, 2016, 17:10 posting](https://translate.google.com/translate?sl=ru&tl=en&js=y&prev=_t&hl=ru&ie=UTF-8&u=http%3A%2F%2Fforums.zyxmon.org%2Fviewtopic.php%3Fp%3D15556%23p15556&edit-text=):

For mipsel (Broadcom) the main differences between **entware-3x** and **entware-ng** are:
1. Newer kernel [3.x versus 2.6.22]
2. glibc system library instead of uclibc [RAM usage <1% different; disk usage is much larger]
3. Possibility to have entware users independent from firmware users
4. Locales independent from firmware
5. Mandatory installation of busybox

zyxmon, 2018-02-07:
Like that.
You can consider Entware-3x as an Entware-ng upgrade.

closed 2018-02-07 by zyxmon