Entware-for-kernel-3x/Entware-ng-3x #61: This is recommended for unification with Entware-3x installations on other devices. The action is not mandatory.

ProBackup-nl, 2018-02-07:
### Package: forum documentation
### Platform: padavan rt-n56u

After reading: https://translate.google.com/translate?sl=ru&tl=en&js=y&prev=_t&hl=ru&ie=UTF-8&u=http%3A%2F%2Fforums.zyxmon.org%2Fviewtopic.php%3Fp%3D15556%23p15556&edit-text=
I am wondering what is meant with:

> This is recommended for unification with Entware-3x installations on other devices. The action is not mandatory.

Is "this" referring to the line before?

```
rm /opt/etc/profile
```

Or after?

```
wget -O - http://entware-3x.zyxmon.org/binaries/mipsel/installer/install_std.sh | /bin/sh
```

In other words: what is optional?

a. `remove .../profile`
or 
b. `run .../install_....sh?`

zyxmon, 2018-02-07:
Online translator interpretaion is not correct here. This refers to busybox.

closed 2018-02-07 by zyxmon

ProBackup-nl, 2018-02-07:
@zyxmon `install_std.sh` installs `entware-opt` which depends upon `busybox`. I don't understand the "not mandatory" part.
Would you mind having a look at https://github.com/ProBackup-nl/Entware-ng-3x/wiki/Install-on-Padavan ? Corrections are welcome, as well as hints to verify that one has not ended up at situation "padavan firmware overwriting entware-3x".

zyxmon, 2018-02-07:
>  I don't understand the "not mandatory" part.

Look here - http://entware-3x.zyxmon.org/binaries/armv7/installer/install_asuswrt.sh 
It is for armv7 & armv8 now. 2 extra packages need to be added for it. Some packages will need additional adjustments when busybox is not installed.
I'll have a look, and probably copy wiki here, if you don't mind.

ProBackup-nl, 2018-02-07:
@zyxmon I don't mind **at all** when you copy (and improve) the wiki.