Entware-for-kernel-3x/Entware-ng-3x #64: Package smcroute not starting

sim2github, 2018-03-07:
### Package: smcroute

- expected behavior: 
run daemon background/foreground
output error
- actual behavior: 
```shell
[etc]# opkg info smcroute
Package: smcroute
Version: 2.0.0-2
Depends: libc, libssp, librt, libpthread
Status: install user installed
Section: net
Architecture: armv7-3x
MD5Sum: c71a17704ec61b80cd8ff51457b7e529
Size: 14448
Filename: smcroute_2.0.0-2_armv7-3x.ipk
Description: SMCRoute is a command line tool to manipulate the multicast routes of the Linux kernel.
Installed-Time: 1520383342

[etc]# smcroute
SMCRoute version 2.0.0
Usage: smcroute [OPTIONS]... [ARGS]...

  -d       Start daemon
  -n       Run daemon in foreground
  -f FILE  File to use instead of default /opt/etc/smcroute.conf
  -k       Kill a running daemon

  -h       This help text
  -D       Debug logging
  -v       Show version and enable verbose logging

  -a ARGS  Add a multicast route
  -r ARGS  Remove a multicast route

  -j ARGS  Join a multicast group
  -l ARGS  Leave a multicast group

     <------------- INBOUND -------------->  <----- OUTBOUND ------>
  -a <IFNAME> <SOURCE-IP> <MULTICAST-GROUP>  <IFNAME> [<IFNAME> ...]
  -r <IFNAME> <SOURCE-IP> <MULTICAST-GROUP>

  -j <IFNAME> <MULTICAST-GROUP>
  -l <IFNAME> <MULTICAST-GROUP>
[etc]#
[etc]# smcroute -d
[etc]# ps | grep smcroute
13655 admin     3088 S    grep smcroute
[etc]# smcroute -n
[etc]# ps | grep smcroute
13748 admin     3088 S    grep smcroute
[etc]# smcroute -D
[etc]# ps | grep smcroute
20551 admin     3088 S    grep smcroute
```
### Platform:
QNAP TS-228
Firmware version: 4.3.4.0486 Build 20180215
CPU: ARMv7 Processor rev 5 (v7l)

zyxmon, 2018-03-07:
You need to edit /opt/etc/smcroute.conf for your device before (auto)starting it with /opt/etc/init.d/Sxxx script

sim2github, 2018-03-07:
This is first that  i have done. 
```sh
[root]# /opt/etc/init.d/S50smcroute start
 Starting smcroute...              failed.
[root]# cat /opt/etc/init.d/S50smcroute
#!/bin/sh

ENABLED=yes
PROCS=smcroute
ARGS="-d"
PREARGS=""                                                                 DESC=$PROCS
PATH=/opt/sbin:/opt/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
. /opt/etc/init.d/rc.func
[root]#
```
Init script do the same as `smcroute -d`

I dont see any log or output so cant understand where is problem.

zyxmon, 2018-03-07:
Probably your config is still invalid. One of the developers started smcroute without problems on his router (not om qnap NAS).

sim2github, 2018-03-07:
Reset config to default - still get silence. If config not valid program must inform me where is mistake. 
Sources [smcroute/src/log.c](https://github.com/troglobit/smcroute/blob/ec5aa5fcb70f1ae8dd3c68be6fa0330d8a2006f0/src/log.c#L54) says that i can find it in syslog or stderr but i don't.

Will try compile from sources.

zyxmon, 2018-03-07:
There is no default config, package comes with some example config to be adjusted. There is no syslog on qnap. You can try to install syslog-ng or similar package.

zyxmon, 2018-03-07:
The syslog-ng shows on TS-128

```
Mar  7 17:51:39 QNAPD1 smcroute[16201]: SMCRoute version 2.0.0
Mar  7 17:51:39 QNAPD1 smcroute[16201]: Failed initializing IPv4 multicast routing API. Error 92: Protocol not available
Mar  7 17:51:39 QNAPD1 smcroute[16201]: Failed initializing IPv6 multicast routing API. Error 92: Protocol not available
Mar  7 17:51:39 QNAPD1 smcroute[16201]: Kernel does not support multicast routing.

```
Your kernel does not support smcroute

closed 2018-03-07 by zyxmon

sim2github, 2018-03-07:
Confirm. :(
Thanks for help. 👍