Entware-for-kernel-3x/Entware-ng-3x #66: New packages: numpy, pandas and matplotlib python3 modules

iacchi, 2018-03-09:
### Packages: python3-numpy, python3-pandas, python3-matplotlib

Please consider adding these three python3 modules to the repository, they would enable analysis of scientific data stored on NAS systems.
I have also tried to compile them myself using pip, but there are compilers missing from the gcc package (specifically arm-openwrt-linux-gnueabi-gcc-ar) and the build fails. There might be other errors after that (matplotlib is missing devel packages for libfreetype and libpng for example), so I think the best option is to provide these in the repository.

### Platform: armv7

I'm planning to install them on a Synonlogy NAS, model DS214+

zyxmon, 2018-03-09:
1. Please reead the wiki - how to compile python packages 
https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules
https://github.com/Entware/Entware/wiki/Using-gcc-%28native-compilation%29
You need to adjust headers URL for your installation
2. Entware-3x & Entware-ng have merged. The new project name is Entware. github locations is
https://github.com/Entware You can easily upgrade to the new project by running `opkg update; opkg upgrade` two times.

closed 2018-03-09 by zyxmon

iacchi, 2018-03-09:
I'm keeping this closed, but I wanted to thank you for the heads up about the project merging - I was wondering why the wiki disappeared from the entware-ng project. If I may give a suggestion, in the initial readme file of both this project and the entware-ng one you should write at the very beginning and in flashy colours that the two projects merged and that people should go to the new project page instead. The new project is pretty badly positoned in Google results, and I didn't even realised it existed despite looking at these things the whole day yesterday!

zyxmon, 2018-03-09:
@iacchi - waiting for your suggestion. Entware merge was announced in different places. But it was not announced on Synology forum. I do not think entware is popular with synology owners.

You can use http://bin.entware.net/Readme.txt as a starting point of your text. It will take sometime to upgrade new wiki.

iacchi, 2018-03-09:
It doesn't need to be anything long or complicated. Maybe something like this, at the beginning of the readme.md file here on github for both projects:

## Entware-ng-3x and Entware-ng merged to become Entware
Entware-ng-3x and Entware-ng decided to join forces and merge the two projects into a new project called [Entware](https://github.com/Entware/Entware). As a result the two projects [Entware-ng-3x](https://github.com/Entware-for-kernel-3x/Entware-ng-3x/) and [Entware-ng](https://github.com/Entware-ng/Entware-ng) will not be maintained anymore. All software and resources will be available at the new project webpage, including the [documentation wiki](https://github.com/Entware/Entware/wiki).